JuliaRobotics/Caesar.jl #46: Handling Array{Any,1}

pvazteixeira, 2017-04-28:
When running `drawdbdirector()`

```julia
ERROR: dont know how to deal with data type=Array{Any,1}
 in #prepcolordepthcloud!#151(::Array{Any,1}, ::Int64, ::Float64, ::Function, ::Int64, ::Array{Any,1}) at /home/pvt/.julia/v0.5/Caesar/src/cloudgraphs/DirectorVisService.jl:111
 in (::Caesar.#kw##prepcolordepthcloud!)(::Array{Any,1}, ::Caesar.#prepcolordepthcloud!, ::Int64, ::Array{Any,1}) at ./<missing>:0
 in #fetchdrawdepthcloudbycvid!#157(::Dict{String,String}, ::CoordinateTransformations.AffineMap{Rotations.Quat{Float64},StaticArrays.SVector{3,Float64}}, ::Function, ::DrakeVisualizer.Visualizer, ::CloudGraphs.CloudGraph, ::Int64, ::Symbol, ::Dict{Any,Any}, ::Dict{String,Any}, ::Symbol) at /home/pvt/.julia/v0.5/Caesar/src/cloudgraphs/DirectorVisService.jl:265
 in (::Caesar.#kw##fetchdrawdepthcloudbycvid!)(::Array{Any,1}, ::Caesar.#fetchdrawdepthcloudbycvid!, ::DrakeVisualizer.Visualizer, ::CloudGraphs.CloudGraph, ::Int64, ::Symbol, ::Dict{Any,Any}, ::Dict{String,Any}, ::Symbol) at ./<missing>:0
 in #fetchdrawposebycvid!#159(::String, ::Dict{String,String}, ::Function, ::DrakeVisualizer.Visualizer, ::CloudGraphs.CloudGraph, ::Int64, ::Dict{Any,Any}, ::Dict{String,Any}) at /home/pvt/.julia/v0.5/Caesar/src/cloudgraphs/DirectorVisService.jl:323
 in (::Caesar.#kw##fetchdrawposebycvid!)(::Array{Any,1}, ::Caesar.#fetchdrawposebycvid!, ::DrakeVisualizer.Visualizer, ::CloudGraphs.CloudGraph, ::Int64, ::Dict{Any,Any}, ::Dict{String,Any}) at ./<missing>:0
 in macro expansion at /home/pvt/.julia/v0.5/Caesar/src/cloudgraphs/DirectorVisService.jl:375 [inlined]
 in macro expansion at /home/pvt/.julia/v0.5/ProgressMeter/src/ProgressMeter.jl:474 [inlined]
 in drawdbsession(::DrakeVisualizer.Visualizer, ::CloudGraphs.CloudGraph, ::Dict{AbstractString,Union{AbstractString,Array{String,1}}}, ::Dict{String,Any}, ::Dict{Any,Any}) at /home/pvt/.julia/v0.5/Caesar/src/cloudgraphs/DirectorVisService.jl:374
 in #drawdbdirector#160(::Dict{AbstractString,Union{AbstractString,Array{String,1}}}, ::Function) at /home/pvt/.julia/v0.5/Caesar/src/cloudgraphs/DirectorVisService.jl:436
 in (::Caesar.#kw##drawdbdirector)(::Array{Any,1}, ::Caesar.#drawdbdirector) at ./<missing>:0
```

Key at fault:  `bDE.mongoKey = "5903a2d56689bc076e31e2d7"`

dehann, 2017-04-30:
Hey Pedro, I think we fixed this on your machine. Could we maybe link the commit sha here when you get a chance please?

pvazteixeira, 2017-05-01:
Fixed here: 3709f36e162a94b985d5c7c056bb53b42650ef53

closed 2017-05-09 by dehann