JuliaRobotics/Caesar.jl #73: Move these functions to CloudGraphs.jl

GearsAD, 2017-11-21:
https://github.com/dehann/Caesar.jl/blob/master/src/cloudgraphs/CloudGraphIntegration.jl

dehann, 2017-11-22:
Thanks, I think CloudGraphs.jl should work at the level of Graphs.ExVertex and CloudVertex, and leave the robotics related calls in either Caesar.jl or RoME.jl. I guess we will see in time how the functions in `CloudGraphIntegration.jl` file should be factored. Thanks!

dehann, 2018-03-03:
Hi, is this still in play?  At what point should we start removing functions from `Caesar/CloudGraphIntegration.jl`?

GearsAD, 2018-09-23:
Hmm, I'm now rethinking. These functions are not db specific - ideally you should be able to do them without knowing that there's a db underneath -> They're technically there because the CloudGraphs layer is too thin. Let's discuss but I think we grow DFG and slowly deprecate the db-specific stuff?

dehann, 2019-03-05:
Agreed, DFG is it I think.  Will add a note to the IIF/DFG project to point here.

dehann, 2019-10-15:
This should be done with DFG v0.4.x

closed 2019-10-15 by dehann