JuliaRobotics/Caesar.jl #91: Convert to native Julia LCM types?

rdeits, 2018-03-11:
@tkoolen has done some nice work over at LCMCore.jl to make it very easy to make native Julia types to match LCM type definitions. It's not quite all the way to `lcm-gen`, but it's pretty easy to use and the resulting types should be extremely performant. I just ported all of the types in `bot_core_lcmtypes` and Drake last night. 

If you're interested, I'm happy to open a PR to show what that would look like here. It should make the LCM interface much more efficient than using PyLCM.

dehann, 2018-04-03:
HI @rdeits , thanks for the heads-up -- sorry I missed your issue before.  Yes, @pvazteixeira and I actually recently spoke about separating out a new package called `CaesarLCMTypes.jl` and use Twan's approach there.  Let me get back to you...

FYI: @GearsAD , @Affie

dehann, 2018-04-15:
see [CaesarLCMTypes.jl](https://github.com/JuliaRobotics/CaesarLCMTypes.jl)

dehann, 2018-04-15:
#96

closed 2018-04-15 by dehann