JuliaRobotics/Caesar.jl #105: Thorough review of CloudGraphs.outneighbors

dehann, 2018-04-25:


dehann, 2018-04-25:
cc @GearsAD , can we always rely on `()-[:DEPENDENCE]-()`?

dehann, 2019-06-14:
HI @GearsAD , this issue is probably under a lot of consideration right now via CloudGraphsDFG?

GearsAD, 2019-06-15:
Hi @dehann,

Okay, really good question. I'd like that:
* Outneighbors returns all nodes in the graph
* It does not return the SESSION node (i.e. the session anchor)
* It does work with multisession prime nodes 

So I've put in that all factor--variable edges are set with :FACTORGRAPH, not :DEPENDENCE (for clarity). We'll maintain that when we do multisession.

E.g.
[image]

Any concerns?

dehann, 2019-10-15:
Provided by DFG v0.4.x

closed 2019-10-15 by dehann