JuliaRobotics/Caesar.jl #107: Solver failing onFullGraphCopy

GearsAD, 2018-04-28:
Initialization seems to be failing intermittently while running real-time. Getting this error when running the solver:

```
===================CONVERT===================
done populating new variables
WARNING: using hack counter for FACTOR uid starting at 100052

=============ENSURE INITIALIZED==============

================MULTI-SESSION================

====================SOLVE====================
Get local copy of graph
fullLocalGraphCopy: 50 nodes in session RoverRover if reqbackendset=true and reqready=true...
Copy all nodes...100%|██████████████████████████████████| Time: 0:00:03
-------------Ensure Initialization-----------
x22 is not initialized, and will do so now...
Looking for cloud out neighbors
cgid = fgl.cgIDs[exVertId] = 600
neinodes = ls(fgl, vsym) = Symbol[:x22x23f1, :x21x22f1]
Looking for cloud out neighbors
cgid = fgl.cgIDs[exVertId] = 627
xfneivarnodes = lsf(fgl, xifct) = Symbol[:x23, :x22]
Looking for cloud out neighbors
cgid = fgl.cgIDs[exVertId] = 1282
xfneivarnodes = lsf(fgl, xifct) = Symbol[:x22, :x21]
 --- Exiting! Deregistering solver with ID '9e994028-749b-4821-9127-c7061a330cb4'!
ERROR: LoadError: BoundsError: attempt to access 0×0 Array{Float64,2} at index [Base.Slice(Base.OneTo(0)), 1]
while loading /home/gearsad/.julia/v0.6/CaesarServer/scripts/runme.jl, in expression starting on line 8
```

The node in question contains the following data
```
bigData: {"isRetrieved":false,"isAvailable":false,"isExistingOnServer":true,"lastSavedTimestamp":"2018-04-28T02:56:13.569","version":"2","dataElements":[]} data: 16,0,32,0,48,0,64,0,80,1,80,2,80,3,88,3,96,0,104,0,122,43,73,110,99,114,101,109,101,110,116,97,108,73,110,102,101,114,101,110,99,101,46,67,111,110,116,105,110,117,111,117,115,77,117,108,116,105,118,97,114,105,97,116,101,128,1,1 exVertexId: 47 label: x23 neighborVertexIDs: 627 packedType: IncrementalInference.PackedVariableNodeData ready: 1
```

dehann, 2018-05-17:
This might be related to previous assumption in creating new nodes.  Old API required the user supply the initial values for a variable node.  The new `addNode!` interface relies on `autoinit` in the `addFactor!` function.  A change was later introduced to a least create the memory for values at `addNode!` time, and assuming that will fix this error.  Lets keep it open and keep testing the in-situ concurrent see if this occurs again.

dehann, 2018-07-12:
https://github.com/JuliaRobotics/IncrementalInference.jl/commit/4ef52d591580b7f2b3c549af7ba998e1d7732e2a#diff-04105ee66372b738f19a2ef200b285f9R931

closed 2018-07-12 by dehann