JuliaRobotics/Caesar.jl #128: How do I draw top down map from SLAM result

dehann, 2018-07-04:
After running `inferOverTree!` , this is how you would draw poses and landmarks:
```julia
using RoMEPlotting

# produce a figure in the julia workspace
pl = drawPoses(fg)

using Gadfly
Gadfly.draw(PDF("/tmp/test.pdf", 20cm,10cm),pl)
# or
drawPosesLandms(fg)
```
cc @keevindoherty

dehann, 2018-07-04:
Gadfly drawing will require two packages, if you don't already have them:
```julia
Base.rm(ENV["HOME"]*"/.julia/lib/v0.6/Compose.ji")
Pkg.add.(["Cairo";"Fontconfig"])
``` 
The next use of Gadfly will recompile compose and allow `draw` to export PDF or PNGs.  Comment here if you require more support for other plotting backends such as `Plots.jl`.

keevindoherty, 2018-07-04:
Running drawPosesLandms(fg) results in an UndefVarError for me: "Colorant not defined"

Could this be an issue with my factor graph, or something else?

dehann, 2018-07-04:
ah,  that would be `Colors` thanks.  What do you get when you run this?
```julia
using Colors
```

dehann, 2018-07-04:
Might be a dependency mistake, let me check quick...

keevindoherty, 2018-07-04:
```julia
julia> pl = drawPosesLandms(fg)
lb = :l1
lbl = string(lb) = "l1"
ERROR: UndefVarError: Colorant not defined
Stacktrace:
 [1] #drawLandms#37(::Int64, ::Int64, ::Int64, ::Symbol, ::Bool, ::Bool, ::Bool, ::String, ::Type{T} where T, ::IncrementalInference.DataLayerAPI, ::RoMEPlotting.#drawLandms, ::IncrementalInference.FactorGraph) at /home/kevin/.julia/v0.6/RoMEPlotting/src/RobotViz.jl:158
 [2] (::RoMEPlotting.#kw##drawLandms)(::Array{Any,1}, ::RoMEPlotting.#drawLandms, ::IncrementalInference.FactorGraph) at ./<missing>:0
 [3] #drawPosesLandms#38(::Int64, ::Int64, ::Int64, ::Symbol, ::Bool, ::Bool, ::Type{T} where T, ::Bool, ::Float64, ::IncrementalInference.DataLayerAPI, ::RoMEPlotting.#drawPosesLandms, ::IncrementalInference.FactorGraph) at /home/kevin/.julia/v0.6/RoMEPlotting/src/RobotViz.jl:181
 [4] drawPosesLandms(::IncrementalInference.FactorGraph) at /home/kevin/.julia/v0.6/RoMEPlotting/src/RobotViz.jl:180
```

dehann, 2018-07-04:
Could you also give me your current `Pkg.status()` please?  I think this might be an upstream issue...

keevindoherty, 2018-07-04:
```
julia> Pkg.status()
11 required packages:
 - Arena                         0.0.1
 - Caesar                        0.1.0
 - Cairo                         0.5.2
 - Fontconfig                    0.1.1
 - Gadfly                        0.7.0
 - GraphPlot                     0.2.0
 - IncrementalInference          0.3.6
 - Mongo                         0.3.0
 - Plots                         0.17.3
 - RoME                          0.1.4
 - RoMEPlotting                  0.0.1
146 additional packages:
 - AbstractFFTs                  0.3.2
 - AssetRegistry                 0.0.2
 - AxisAlgorithms                0.2.0
 - AxisArrays                    0.2.1
 - BinDeps                       0.8.8
 - BinaryProvider                0.3.3
 - Blosc                         0.5.0
 - BufferedStreams               0.4.0
 - CMakeWrapper                  0.1.0
 - Calculus                      0.4.0
 - CatIndices                    0.1.0
 - CategoricalArrays             0.3.10
 - CloudGraphs                   0.0.2
 - CodecZlib                     0.4.3
 - Codecs                        0.4.0
 - ColorTypes                    0.6.7
 - ColorVectorSpace              0.5.2
 - Colors                        0.8.2
 - CommonSubexpressions          0.1.0
 - Compat                        0.69.0
 - Compose                       0.6.0
 - ComputationalResources        0.2.0
 - Contour                       0.4.0
 - CoordinateTransformations     0.4.1
 - CoupledFields                 0.0.1
 - CustomUnitRanges              0.1.0
 - DataArrays                    0.7.0
 - DataFrames                    0.11.6
 - DataStreams                   0.3.6
 - DataStructures                0.8.3
 - DiffBase                      0.3.2
 - DiffEqDiffTools               0.4.1
 - DiffResults                   0.0.3
 - DiffRules                     0.0.6
 - Distances                     0.6.0
 - Distributions                 0.15.0
 - DocStringExtensions           0.4.5
 - DrakeVisualizer               0.3.0
 - DualNumbers                   0.4.0
 - FFTViews                      0.1.0
 - FFTW                          0.0.4
 - FileIO                        0.9.1
 - FixedPointNumbers             0.4.6
 - ForwardDiff                   0.7.5
 - FunctionalCollections         0.3.2
 - GR                            0.31.0
 - GeometryTypes                 0.4.6
 - Graphics                      0.3.0
 - Graphs                        0.9.0
 - Gtk                           0.14.0
 - GtkReactive                   0.4.1
 - HDF5                          0.9.3
 - Hexagons                      0.1.0
 - Hiccup                        0.1.1
 - HttpCommon                    0.4.0
[... 67 lines cut ...]
 - Requests                      0.5.1
 - Requires                      0.4.4
 - Rmath                         0.4.0
 - Rotations                     0.7.1
 - RoundingIntegers              0.0.3
 - SHA                           0.5.7
 - SIUnits                       0.1.0
 - ShowItLikeYouBuildIt          0.2.0
 - Showoff                       0.2.0
 - SimpleTraits                  0.6.0
 - SortingAlgorithms             0.2.1
 - SpecialFunctions              0.6.0
 - StaticArrays                  0.7.2
 - StatsBase                     0.23.1
 - StatsFuns                     0.6.1
 - TexExtensions                 0.1.0
 - TiledIteration                0.1.0
 - TranscodingStreams            0.5.2
 - TransformUtils                0.1.0
 - URIParser                     0.3.1
 - WeakRefStrings                0.4.7
 - WebIO                         0.2.4
 - WebSockets                    0.5.0
 - WoodburyMatrices              0.3.0
```

dehann, 2018-07-04:
looks like this was partly a versionitus problem, `RoME v0.1.5` should be tagged soon.  The latest from `RoMEPlotting` is also required -- `v0.0.2` coming soon.

dehann, 2018-07-05:
see: http://www.juliarobotics.org/Caesar.jl/latest/tut_hexagonal2d.html

closed 2018-07-05 by dehann