JuliaRobotics/Caesar.jl #133: How can you create a contour plot of the current state estimate?

keevindoherty, 2018-07-09:
Current examples show particle distribution, but e.g. RoME.jl readme shows contour plot of distribution

dehann, 2018-07-10:
Please see new documentation here, and list further issues:
http://www.juliarobotics.org/Caesar.jl/latest/arena_visualizations.html#Density-Contour-Map-1

One note is that the API for PriorPose2/Point2 is busy standardizing, so there might be a slight differences in how I use them from what you have seen before.  You should also see deprecation warnings if you are using the older API on new versions of RoME or Caesar.

Thanks for opening the issue.

closed 2018-07-10 by dehann

dehann, 2018-07-10:
see more examples here
http://www.juliarobotics.org/Caesar.jl/latest/tut_slamedonut.html#Inference-and-Visualizations-1