JuliaRobotics/Caesar.jl #140: Does Caesar provide a way perform parametric Inference?

keevindoherty, 2018-07-13:
For example, if I have all Gaussian factors, is there an easy way within Caesar to see what the uni-modal solution would be (e.g. as if I were using iSAM2)

dehann, 2018-07-17:
Not yet, but pretty easy to add -- I just have to get round to doing it.  Several people have asked for it though.  Let me get this properly into the triage.

dehann, 2018-07-17:
Note on version numbers. `v0.2.0` will be associated with Julia 0.7 which is currently at beta-2 stage and should be coming around soon.

dehann, 2019-02-15:
cc @Affie

Affie, 2022-05-09:
Parametric batch solve currently works on variable type Pose2, Point2, and Continous Scalar.
More is planned with a full upgrade to Manifolds.jl

Use `IIF.solveGraphParametric!(fg)`

Affie, 2022-06-28:
Parametric Batch Solve should be able to work on all lie groups supported by Manifolds.jl (tested on Pose3). It currently does have a performance issue though.
Please log an issue if you encounter problems.

dehann, 2022-06-28:
This is part of IncrementalInference.jl v0.29 following Manifolds.jl v0.8.11

closed 2022-06-28 by dehann