JuliaRobotics/Caesar.jl #141: How to use the multi-hypothesis feature?

keevindoherty, 2018-07-16:
Is there a simple example making use of the multi-hypothesis API?

dehann, 2018-07-17:
lets say you have an ambiguous `pp = Pose2Point2BearingRange(...)` to one of two landmarks, you can update the `addFactor!` part as follows:
```julia
addFactor!(fg, [:x1;:l1;:l2], pp, multihypo=[1.0;0.5;0.5])
```
Assuming you are working from pose `:x1` and ambiguity exists between `:l1 / :l2`.  The intended use of the `multihypo` keyword is a simplification for bimodal factors. (Future versions will implicitly allow n-modal definition via the 'easy API').  In this example the ambiguity is taken as 0.5/0.5 split between `:l1 / :l2`.  The syntax requires you to specify the 1.0 certainty in `:x1`.  You should be able to use this ambiguity definition with any factor in a bimodal fashion.  Note this interface is a convenience method and the internals of multimodal-iSAM allow for far greater complexity.

Should probably add an example in the documentation.

GearsAD, 2018-11-12:
+1 for doc example

dehann, 2019-05-05:
will be added once the fractional null-hypothesis update in IIF is completed -- JuliaRobotics/IncrementalInference.jl#248

dehann, 2020-10-27:
really need to document for JuliaRobotics/RoME.jl#342

dehann, 2020-12-09:
https://juliarobotics.org/Caesar.jl/latest/concepts/dataassociation/

closed 2020-12-09 by dehann