JuliaRobotics/Caesar.jl #152: Deserialization issues - have packed structures changed?

GearsAD, 2018-09-20:
You may have mentioned this, but wanted to confirm. I'm seeing deserialization issues when solving, if I upgrade everything to latest and reinsert the graphs, would that resolve the issue?

ERROR: LoadError: MethodError: Cannot `convert` an object of type Type{RoME.PackedPriorPose2} to an object of type IncrementalInference.FunctorInferenceType
This may have arisen from a call to the constructor IncrementalInference.FunctorInferenceType(...),
since type constructors fall back to convert methods.
Stacktrace:
 [1] decodePackedType(::IncrementalInference.GenericFunctionNodeData{RoME.PackedPriorPose2,String}, ::String) at /home/gearsad/.julia/v0.6/IncrementalInference/src/DispatchPackedConversions.jl:293
 [2] unpackNeoNodeData2UsrType(::CloudGraphs.CloudGraph, ::Neo4j.Node) at /home/gearsad/.julia/v0.6/CloudGraphs/src/CloudGraphs.jl:135
 [3] neoNode2CloudVertex(::CloudGraphs.CloudGraph, ::Neo4j.Node) at /home/gearsad/.julia/v0.6/CloudGraphs/src/CloudGraphs.jl:144
 [4] get_vertex(::CloudGraphs.CloudGraph, ::Int64, ::Bool) at /home/gearsad/.julia/v0.6/CloudGraphs/src/CloudGraphs.jl:217
 [5] macro expansion at /home/gearsad/.julia/v0.6/Caesar/src/cloudgraphs/CloudGraphIntegration.jl:581 [inlined]
 [6] macro expansion at /home/gearsad/.julia/v0.6/ProgressMeter/src/ProgressMeter.jl:483 [inlined]
 [7] copyAllNodes!(::IncrementalInference.FactorGraph, ::Dict{Int64,CloudGraphs.CloudVertex}, ::Array{Tuple{Int64,Int64,Symbol},1}) at /home/gearsad/.julia/v0.6/Caesar/src/cloudgraphs/CloudGraphIntegration.jl:580
 [8] copyGraphNodesEdges!(::IncrementalInference.FactorGraph, ::Array{Tuple{Int64,Int64,Symbol},1}) at /home/gearsad/.julia/v0.6/Caesar/src/cloudgraphs/CloudGraphIntegration.jl:607
 [9] #fullLocalGraphCopy!#32(::Bool, ::Bool, ::Function, ::IncrementalInference.FactorGraph) at /home/gearsad/.julia/v0.6/Caesar/src/cloudgraphs/CloudGraphIntegration.jl:652
 [10] runSlamInDbOnSession(::Caesar.CaesarConfig, ::CloudGraphs.CloudGraph, ::String, ::String, ::String, ::Int64, ::Bool, ::Caesar.SolverStatus) at /home/gearsad/.julia/v0.6/Caesar/src/cloudgraphs/slamindb.jl:76
 [11] runCaesarOnSessionWithLogging(::CaesarServer.SystemConfig, ::CloudGraphs.CloudGraph, ::String, ::String, ::String, ::Int64, ::Bool, ::Caesar.SolverStatus, ::Bool) at /home/gearsad/.julia/v0.6/CaesarServer/src/services/CaesarLogService.jl:65
 [12] startCaesarServer() at /home/gearsad/.julia/v0.6/CaesarServer/src/CaesarServer.jl:126
 [13] include_from_node1(::String) at ./loading.jl:576
 [14] include(::String) at ./sysimg.jl:14
 [15] process_options(::Base.JLOptions) at ./client.jl:305
 [16] _start() at ./client.jl:371

dehann, 2018-09-23:
Hi, I would have expected this to work.  Let me do some digging...

GearsAD, 2018-09-23:
Fixed in https://github.com/JuliaRobotics/IncrementalInference.jl/pull/163

closed 2018-09-23 by GearsAD