JuliaRobotics/Caesar.jl #168: suggestion for request syntax

pvazteixeira, 2018-10-21:
multi-lang requests should, at the higher level, just have `request` and `payload` fields, e.g.:

```json
{
  "request": "addFactor",
  "payload": {
    "factorType": "Prior",
    "measurements": {
      ...
    },
    "variables": [ "x0"]
  }
}
```

See also: [graff_cpp/pull/14](https://github.com/pvazteixeira/graff_cpp/pull/14/files)

GearsAD, 2018-10-27:
Done - it's now in the feature/multilangzmq branch and with the 0.7 updates, e.g.: https://github.com/JuliaRobotics/Caesar.jl/blob/feature/multilangzmq/test/multilangzmq/zmqInternal.jl#L31-L32

GearsAD, 2018-10-27:
@pvazteixeira are you ok to upgrade to 0.7? We can then merge the branch and all switch over.

closed 2018-10-28 by dehann