JuliaRobotics/Caesar.jl #210: Compile Static .so libraries for multi-language or executable use

dehann, 2018-11-25:
see:
- https://juliacomputing.com/blog/2016/02/09/static-julia.html
- https://github.com/JuliaLang/PackageCompiler.jl
- https://discourse.julialang.org/t/compiler-work-priorities/17623

dehann, 2018-11-30:
xref documentation:
http://www.juliarobotics.org/Caesar.jl/latest/concepts/multilang/#Static,-Shared-Object-.so-Compilation-1

dehann, 2020-05-22:
will use PackageCompiler.jl

Affie, 2022-04-16:
https://discourse.julialang.org/t/successful-static-compilation-of-julia-code-for-use-in-production/79318

dehann, 2022-04-19:
Oh very cool!  I noticed StaticCompiler the other day but wasn't exactly sure how it fit in alongside PackageCompiler.  The Discourse post answers that :-)  Thanks for sharing!  Want to dig in when I get a chance.

> The [StaticCompiler 90](https://github.com/tshort/StaticCompiler.jl) package was recently registered. This post records a successful experiment to statically compile a piece of Julia code into a small .so library on Linux, which is then loaded from Python and used in training of a deep learning model.