JuliaRobotics/Caesar.jl #251: Labels tags or ids

dehann, 2019-02-12:


dehann, 2019-02-12:
When do we use labels and tags for a node?  And why do softtypes have labels too?

dehann, 2019-02-12:
Graphs.jl has vert .index and vert .label 
Then I added label to addVariable for db stuff. Then I added softtypes. Then I added labels to softtypes for some reason, and now we confused

dehann, 2019-02-12:
Johan, I think punchline is labels are so useful in db that we want them in memory too — it was never built that way ... clearly we have to do a better job of that.  We likely won’t go as fancy as the db, but at least something.  You are rightly asking for the labeltags to be available and persistent in memory

dehann, 2019-02-12:
Cc @GearsAD @Affie

Affie, 2019-02-12:
for in memory factor graphs it would be handy to distinguish between different types, eg. poses, landmarks etc.

GearsAD, 2019-02-12:
As per Hangout conversation:

- Labels in memory should be 1st class citizens, as they are in DB. This is an issue we need to address, probably in CloudGraphs.
- Let's call the x1/l1 marker an ID or a tag to differentiate

dehann, 2019-02-12:
+1 for :x1/:l2/:p4 is a tag

Affie, 2019-02-12:
if you call `addNode!(fg, :l1, Pose2, labels=["LANDMARK";])` on local fg, what happens to the label currently? The equivalent Graff call `addVariable(...` keeps it.

GearsAD, 2019-02-12:
When you work with in-memory vertices, you'll have:
- vert.index (internal location ID)
- vert.label = x1/x2/etc = node tag
- vert.attributes["data"] = data
- vert.attributes["labels"] = database labels.

Is that correct and good with everyone?

dehann, 2019-02-12:
Joha, you were asking about Pose2Pose2 overwriting Variable labels?

Affie, 2019-02-12:
No, that's something else. Still debugging.

dehann, 2019-02-12:
Okay, will do Pose2Pise2 separately

dehann, 2019-02-12:
We also have to figure out if softtype has any use.  I used to think (for lack of insight) that pose 2 is likely a “POSE” — that is definitely not true.  So I’m not sure if softtype.label is useful.

My suggestion is we leave the softtype data field there but ignore it and don’t use it.  There only use getData(vert) labels or vert.attributes[“labels”]

Affie, 2019-02-12:
Do you populate the softtype.label field, if not hard coded in type constructor?

dehann, 2019-02-12:
Probably not.  I think there is one step at node creation where all the labels get merged and stored.  Will have to check precisely what happens there

Affie, 2019-02-12:
For Arena I would like to use something like:
```julia
if in("POSE", nodelabels)
	groupsym = :poses #add to poses group
elseif in("LANDMARK", nodelabels)
	groupsym = :landmarks #add to landmarks group
end
```

Affie, 2019-02-13:
Possibly related in GraffSDK for new variable creation with different labels:
GraffSDK Node:
 - ID: 21955
 - Label: x10
 - Graph Labels: ["**POSE**", "VARIABLE", "...

GraffSDK Node:
 - ID: 21827
 - Label: l0
 - Graph Labels: ["**POSE**", "VARIABLE", "**LANDMARK**", "...

Maybe fixed in https://github.com/JuliaRobotics/RoME.jl/issues/151

dehann, 2019-02-14:
Yes, the intention was to fix this with JuliaRobotics/RoME.jl#151 .

GearsAD, 2019-03-05:
From my understanding this is now fixed, and we're going ahead with label = x1, tags = [VARIABLE, SOMETHING], and ID = 74112. I've updated Graff and API, can close?

dehann, 2019-03-10:
Keeping open until this has been added to Caesar Documentation.  Also added to new Docs kanban.

dehann, 2019-10-15:
its in the wiki.

closed 2019-10-15 by dehann