JuliaRobotics/Caesar.jl #252: Possible loss of factor neighbor information

dehann, 2019-02-12:
Here is a factor graph where 
ls(fg, :l11)
1-element Array{Symbol,1}:
 :x109l11f1

But

ls(fg, :x3)
5-element Array{Symbol,1}:
 :x2x3f1
 :x3l11f1
 :x3l12f1
 :x3l13f1
 :x3x4f1

How is the factor graph being built?

dehann, 2019-02-12:
Maybe try
vert = getVert(fg, :l11)
Graphs.out_neighbors(vert, fg.g) ?

Affie, 2019-02-12:
depending on type of vertex, 
```julia
addNode!(fg, nnode, Pose2, labels=[poselabel])
addFactor!(fg, [pnode;nnode], pp )

addNode!(fg, nnode, Point2, labels=["LANDMARK"])
addFactor!(fg, [pnode;nnode], p2br)
```

Affie, 2019-02-12:
```julia
Graphs.TargetIterator{GenericIncidenceList{ExVertex,Edge{ExVertex},Dict{Int64,ExVertex},Dict{Int64,Array{Edge{ExVertex},1}}},Array{Edge{ExVertex},1}}(Undirected Graph (515 vertices, 525 edges), Edge{ExVertex}[edge [521]: vertex [510] "l11" -- vertex [511] "x109l11f1"])
```

Affie, 2019-02-12:
I look like this happens if a new node is made that already exists. Should we add a warning?

dehann, 2019-02-12:
Just making sure -- the issue is that variable tags (:l1) was being reused?

dehann, 2019-02-12:
there should be a warning for stepping on the same tag name

GearsAD, 2019-02-12:
+1, we need checks on duplication etc when building graphs in memory

dehann, 2019-02-14:
Done

closed 2019-02-14 by dehann