JuliaRobotics/Caesar.jl #256: Multihypo seems to be doing the wrong thing on 1D scalar example?

keevindoherty, 2019-02-13:
It seems like the example below produces some funky marginals...
I'm on Julia 0.6.3

Package versions:
- IIF: 0.3.8+ master
- RoME 0.1.6+ master
- RoMEPlotting 0.0.1+ master
- Distributions 0.15.0
- KernelDensityEstimate 0.3.1+ master

```julia
using IncrementalInference, RoME, RoMEPlotting, Distributions, KernelDensityEstimate

lm_prior_noise = 0.01
meas_noise = 0.25
odom_noise = 0.1
n_samples = 200

# initialize mean landmark locations
l0 = 0.0
l1 = 1.0
l2 = 4.0

# "Ground-truth" robot poses
x0 = 0.0
x1 = 1.0
x2 = 2.0
x3 = 4.0

# Initialize empty factor graph
fg = initfg()

# Place strong prior on locations of three "doors"
addNode!(fg, Symbol("l0"), ContinuousScalar, N=n_samples)
addFactor!(fg, [:l0], RoME.Prior(Normal(l0, lm_prior_noise)))

addNode!(fg, Symbol("l1"), ContinuousScalar, N=n_samples)
addFactor!(fg, [:l1], RoME.Prior(Normal(l1, lm_prior_noise)))

addNode!(fg, Symbol("l2"), ContinuousScalar, N=n_samples)
addFactor!(fg, [:l2], RoME.Prior(Normal(l2, lm_prior_noise)))

# Add first pose
addNode!(fg, Symbol("x0"), ContinuousScalar, N=n_samples)

# Make first "door" measurement
addFactor!(fg, [:x0; :l0; :l1; :l2], LinearConditional(Normal(0, meas_noise)), multihypo=[1.0; 1.0/3.0; 1.0/3.0; 1.0/3.0])

# Add second pose
addNode!(fg, Symbol("x1"), ContinuousScalar, N=n_samples)

# # Gaussian transition model
addFactor!(fg, [:x0; :x1], LinearConditional(Normal(1, odom_noise)))

# # Make second "door" measurement
addFactor!(fg, [:x1; :l0; :l1; :l2], LinearConditional(Normal(0, meas_noise)), multihypo=[1.0; 1.0/3.0; 1.0/3.0; 1.0/3.0])

# # Add third pose
addNode!(fg, Symbol("x2"), ContinuousScalar, N=n_samples)

# # Gaussian transition model
addFactor!(fg, [:x1; :x2], LinearConditional(Normal(1, odom_noise)))

# Add fourth pose
addNode!(fg, :x3, ContinuousScalar, N=n_samples)

# Add odometry transition and new landmark sighting
addFactor!(fg, [:x2, :x3], LinearConditional(Normal(2, odom_noise)))
addFactor!(fg, [:x3; :l0; :l1; :l2], LinearConditional(Normal(0, meas_noise)), multihypo=[1.0; 1.0/3.0; 1.0/3.0; 1.0/3.0])

ensureAllInitialized!(fg)
tree = wipeBuildNewTree!(fg)
inferOverTree!(fg, tree, N=n_samples)
```

dehann, 2019-02-13:
Hi Kevin, got it and thanks for posting — let me dig in and will report here what I find.

dehann, 2019-02-14:
possible connection with---pending cross-examination---with JuliaRobotics/IncrementalInference.jl#162

dehann, 2019-02-14:
Must also confirm this is not related to JuliaRobotics/IncrementalInference.jl#57

dehann, 2019-02-15:
influences #141

dehann, 2019-02-19:
Just a heads up:  I am making progress on another issue in IncrementalInference that may well influence the one described here.  Will link updates as they become available.

dehann, 2019-02-19:
Limited progress with partial fix with JuliaRobotics/IncrementalInference.jl#223, but likely found a route more general solutions (ie fixing a new bug found)

dehann, 2019-02-19:
FYI, wip https://github.com/JuliaRobotics/RoME.jl/blob/31cfc065480f0b4d7edb84298a3c5abd6e3526ef/test/testMultiHypo3Door.jl#L1

dehann, 2019-03-18:
moving this issue to IncrementalInference.jl