JuliaRobotics/Caesar.jl #274: Managing MIME types for bigData storage

dehann, 2019-03-10:
1. Need an example of how to use jpeg vs. png vs. pointcloud.
2. Need second example of how to create your own mime type.
3. Need an example how to ignore mime type and just dump binary or string data.

dehann, 2019-03-12:
This is to allows standard feature in Arena.jl etc to work if MIME type is configured properly.

cc @Affie , @GearsAD

dehann, 2019-03-12:
xref design discussion: https://github.com/JuliaRobotics/Caesar.jl/wiki/Standard-MIME-type-for-bigData----system-wide-integration

dehann, 2020-05-22:
Hi @Affie , just a refresher on this issue -- I think a lot of this has been done in the recent DFG data interface work?

dehann, 2020-08-12:
xref JuliaRobotics/DistrubtedFactorGraphs.jl#579

Affie, 2020-08-12:
We can currently do this one:

> Need an example how to ignore mime type and just dump binary or string data.

I think we should support a few other natively and provide examples of how to add more.
So agree with fist post.

closed 2020-08-20 by dehann