JuliaRobotics/Caesar.jl #279: Test for connectedness when solving

GearsAD, 2019-03-12:
https://github.com/GearsAD/GraffSDK.jl/issues/103

dehann, 2019-03-13:
Also assigning Sam, since I will do the fix on Caesar side and then we can just hash it out together at some point in the Caesar v0.3.1 cycle.

dehann, 2019-05-05:
Moving to next patch release to allow for new JuliaRegistrator tagging approach and other bug fixes.  This function is simultaneously being introduced as part of the DFG upgrade -- tentatively scheduled for IIF v0.7.x cycle (currently in progress).

dehann, 2019-06-25:
DFG introduces the ability to test for connectedness, must now be integrated with solver.

dehann, 2019-06-25:
affects #129

Affie, 2020-08-21:
Is a check needed now that IIF can solve disconnected graphs?