JuliaRobotics/Caesar.jl #281: CRUD for variables and factors

GearsAD, 2019-03-18:


dehann, 2019-03-21:
xref: https://github.com/JuliaRobotics/IncrementalInference.jl/projects/1#card-19143587

dehann, 2019-03-28:
What's in the code so far:
* [`IIF.addVariable!`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/2273aed73ad76c14d893d75b9fd7f37339bcbe2e/src/FactorGraph01.jl#L287)
* [`IIF.deleteVariable!`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/1019ee6dfa28f4bc74309d003f9fe8f40720c9ab/src/FactorGraph01.jl#L786)
* [`IIF.addFactor!`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/2273aed73ad76c14d893d75b9fd7f37339bcbe2e/src/FactorGraph01.jl#L679)
* [`IIF.deleteFactor!`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/1019ee6dfa28f4bc74309d003f9fe8f40720c9ab/src/FactorGraph01.jl#L755)

dehann, 2019-05-15:
Recently added `Graphs.delete_vertex!`.  Should be available in Graphs.jl v0.10.3.

dehann, 2019-08-07:
I think this is now working because of JuliaRobotics/DistributedFactorGraphs.jl, right?

dehann, 2019-10-15:
Moved to DFG and likely resolved at this point.  xref JuliaRobotics/DistributedFactorGraphs.jl#134

closed 2019-10-15 by dehann