JuliaRobotics/Caesar.jl #325: tag v0.3.1

dehann, 2019-05-05:
@JuliaRegistrator register()

JuliaRegistrator, 2019-05-05:
Registration pull request created: [JuliaRegistries/General/560](https://github.com/JuliaRegistries/General/pull/560)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.
    
This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.3.1 -m "<description of version>" 0296a34c984abc8633b73737dbf524296e353de6
git push origin v0.3.1
```

dehann, 2019-05-05:
https://github.com/JuliaRobotics/Caesar.jl/releases/tag/v0.3.1

closed 2019-05-05 by dehann