JuliaRobotics/Caesar.jl #335: Range only example failing unexpectedly (maybe solver update bug)

dehann, 2019-05-31:
The Ranges only example is not producing the right solver output:
http://www.juliarobotics.org/Caesar.jl/latest/examples/basic_slamedonut/

dehann, 2019-07-12:
This should be fixed.  Also look out for new IIF v0.7.3 coming soon

closed 2019-07-12 by dehann

lemauee, 2020-11-02:
Hello,

I tried to compute the range only example today using the RangesExample.jl script and (because that did not yield the expected results) also using the step-by-step approach using the REPL. Both ways yield not totally wrong, but still unsatisfying results:
[image]
The futile attempt of increasing the number of kernels to 1000 didn't help either ;)
[image]

Rolling back to IIF v0.5.7 using `add IncrementalInference@v0.5.7` like the docs suggest sadly didn't work smoothely (RoME precompilation breaks, maybe some other modules also need to be rolled back manually?)
```julia
LoadError: Failed to precompile RoME [91fb55c2-4c03-5a59-ba21-f4ea956187b8] to /home/leopold/.julia/compiled/v1.5/RoME/VVxXB_FMZ9e.ji.
in expression starting at /home/leopold/workspace/Masterarbeit/svn/julia/mmiSAM/examples/RangesExample.jl:6
error(::String) at error.jl:33
compilecache(::Base.PkgId, ::String) at loading.jl:1305
_require(::Base.PkgId) at loading.jl:1030
require(::Base.PkgId) at loading.jl:928
require(::Module, ::Symbol) at loading.jl:923
include_string(::Function, ::Module, ::String, ::String) at loading.jl:1088
```
Other, simpler canonical range-only examples work.

Best,
Leo

Affie, 2020-11-05:
It looks like range only does not work well with `useMsgLikelihoods=true` yet. So, for now, it may be better to use `useMsgLikelihoods=false`

lemauee, 2020-11-05:
Hi, 

using `getSolverParams(fg).useMsgLikelihoods = false` gives me this result:
[image]
Which looks more consistent in itself, but still has an "orientation offset" compared to the GT presented in RangesExample.jl. Is this to be expected with this example? Is the a programmatic way to check how good which contraints (factors) are satisfied? (How to check the residuals?, Something like a cost function in gaussian SLAM) 

Is there any documentation what `useMsgLikelihoods` (or other not-so-self-explanatory ;)) solver settings do?

OT: Do you have any idea why i get the blue rectangles in these plots? Turning off the contours or points did not help.

Best,
Leo

Affie, 2020-11-05:
I think @dehann can answer you on this.

dehann, 2020-11-06:
Hi Leo, I'm going to look at this more via issue JuliaRobotics/IncrementalInference.jl#1010 .  This issue 335 is likely not relevant any more (given major improvements to the solver over the past two years).  There are a few fine grain details regarding treeinit and differential info and factor selection which is not totally automated yet.  A range-only case is good making sure low rank information is properly handled.  I will work with this some and post in IIF 1010, might take a day or two for me to find the best place to pin down the details.

dehann, 2020-11-12:
Hi @lemauee , just note a documentation update on plotting, this may help visualize with some of the more recent functionality:
https://juliarobotics.org/Caesar.jl/latest/concepts/2d_plotting/

lemauee, 2020-11-12:
Thanks a lot, especially the documentation of the Local Graph Product Plot will be useful for me!