JuliaRobotics/Caesar.jl #370: dyn pts not connected to sas factor seem to freeze the solve

mc2922, 2019-07-13:
Ref example script : testDynPtExFail.jl

For size of 1 or 2 cliqs: solve completes, e.g. 
[image]
-> this is with (3) dyn pts ahead of the sas factor in the fg
-> in the script, use the following variables from line 18-20
windowlen = 8; #total variables in fg
saswindow = 5; #size of sas factor
sasstart = 4; #starting id of sas factor

For size of 3+ cliqs, solve freezes (for me at state machine iter 100):
[image]
-> this is with (4) dyn pts ahead of the sas factor in the fg
-> lines 18-20
windowlen = 9;
saswindow = 5;
sasstart = 5;

dehann, 2019-07-14:
There is definitely a bug in tree autoinit solve (as with this case) when there are sibling cliques.  The issue is that CSM is too eagerly blocking both siblings for different reasons.  After :x1 clique finishes (as indicated here) then :x7 should unblock in the priority order and complete it's upsolve.  Green means trying to initialize numerical values.  I'm working on this right now.

dehann, 2019-07-14:
Solver should be (but clearly isn't) following the progression described in JuliaRobotics/IncrementalInference.jl#344

dehann, 2019-07-14:
[image]

dehann, 2019-07-14:
No, there is a deeper issue here.  This example shows it quite well actually:  The two siblings have now common separator variables.  The current CSM code assumed there would be, which is was a mistake.  

The solution is that the factors in the parent clique must also be used (where possible) to transfer information between siblings during the initialization process.  This makes total sense but I didn't see it that way before.

What I initially though was the issue is still an issue, but this example shows a second issue.  When transferring info from upsolved clique (:x1)'s :x3 variable to the right hand clique must use the :x3x4f1 odometry factor in the parent for a valid intersect with clique (:x7)'s separators.

dehann, 2019-07-14:
This other example shows the (what I initially thought) issue where (:x2) should solve first and then (:x4).  This example does not require parent factors to be used where transferring between siblings.

[image]

dehann, 2019-07-14:
Another example with connections between odometry chain on `:x*` and  `ls(fg, :l1) = Symbol[:x6l1f1, :x0l1f1]`, and one prior: `lsfPriors(fg) = Symbol[:x0f1]`
[image]

mc2922, 2019-08-02:
Update on this: adding two factors freezes the solve at this point: 
[image]

mc2922, 2019-08-02:
factor graph for two factors within an odo chain: 
[image]

please see example code here 
https://github.com/JuliaRobotics/Caesar.jl/blob/dynSAS/examples/marine/asv/kayaks/testDynPt2factors.jl

mc2922, 2019-08-03:
Resolved by increasing the solver limit
getSolverParams(fg).limititers=200

[image]

dehann, 2019-08-03:
great thanks

dehann, 2019-08-03:
also requires the parent sharing upgrade to the CSM code (used in the previous figure).

dehann, 2019-08-22:
original issue resolved but more working on tree based initialization required.

closed 2019-08-22 by dehann