JuliaRobotics/Caesar.jl #371: manual init bypassing freeze-up but initializes variables at origin

mc2922, 2019-07-13:
Re: issue [370](https://github.com/JuliaRobotics/Caesar.jl/issues/370)
See script: [testDynPtExManual](https://github.com/JuliaRobotics/Caesar.jl/blob/dynSAS/examples/marine/asv/kayaks/testDynPtExManual.jl)

Same fg that freezes, now added manualinit at line 70. Solve completing:
[image]

but variables are initialized at 0:
[image]
The sas-factor here is x5-x9

dehann, 2019-07-14:
Hi, thanks for posting -- let me have a closer look.  My feeling is there is autoinit bug that is at least partially related to this "numerical" error bug.  Example is very helpful in finding and fixing, thanks.

dehann, 2019-07-15:
HI @mc2922 , is it possible to plot the ground truth values perhaps please?

mc2922, 2019-07-15:
Yup, Line 95 should be plotting ground truth? Let me check it

dehann, 2019-08-02:
Update with latest IIF master, looks like it is working perhaps.  @mc2922 if you could please confirm these results?  I'm having a little trouble with the ground truth still -- anyway here is what I get now:
[image]
[image]

dehann, 2019-08-02:
I think the manual init is still failing, but autoinit is now performing better.  Here is manual init result:
[image]

mc2922, 2019-08-02:
Yes, freeze is resolved, now the tree-init completes with the following result: 
[image]

mc2922, 2019-08-02:
Seeing the same issue with manual init, points are at the origin 
[image]

dehann, 2019-10-10:
probably fixed, but will retest for v0.4.3