JuliaRobotics/Caesar.jl #374: odd man out behavior?

mc2922, 2019-07-18:
script: examples/marine/asv/kayaks/smallRangeError.jl

Adding an odometry chain: x1  - x9
Priors added on x1 and x9
Range factors added on x1, x4 and x7 to beacon l1

Resulting in x4 being an odd man out of the chain?

[image]
[image]

Check the range sampler on :x4, looks reasonable
[image]

bit of looking the marginals on the tree:
[image]

leaf clique with frontal l1 passes up to x1 this msg.
[image]

and up passes to x4 this msg:
[image]

because x1 has the prior?

dehann, 2019-07-19:
Right, so looks like this example is using the tree initialization (as we'd like to).  The up message on :x1 from clique (:l1) is based on the prior, however, only a partial solve is for :x4 and :x7 can be done at the leaff clique as a first step using the range measurements in that clique.  The CSM logic must either accept this for upsolve, or first try provide more information with downward messages before accepting values as the final upsolve result on (:l1).

Also note that just above :x4 is a frontal in (:x4) and should be set there while also considering the odometry factors to :x5, etc also.  Thanks very much for the example and the input -- it defintely helps isolate and guide the CSM fixes going on.  Just FYI:  ongoing work under issues such as JuliaRobotics/IncrementalInference.jl#344.

dehann, 2019-07-19:
The fact that :x4 is so way out after downsolve definitely means all the information from factors connected to :x4 are not making it into the final result for some reason.  And, would suggest there is a sequencing issue with CSM maybe skipping a step as you suggested yesterday.

dehann, 2019-08-03:
I'm getting a different tree for some reason??
[image]

mc2922, 2019-08-03:
Latest IIF result: 
[image]

mc2922, 2019-08-03:
Fixed by using odometry or prior based autoinit

[image]

dehann, 2019-10-10:
Likely fixed in IIF v0.7.7 or greater.

closed 2019-10-10 by dehann