JuliaRobotics/Caesar.jl #378: Precompiling Ceasar fails

ufechner7, 2019-07-20:
Since about a month I cannot precompile Ceasar any longer. Using Julia 1.1 on Ubuntu Linux.
Any idea?
```Precompiling Caesar
[ Info: Precompiling Caesar [62eebf14-49bc-5f46-9df9-f7b7ef379406]
[ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
[ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
[ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
WARNING: Method definition (::Type{DistributedFactorGraphs.GenericFunctionNodeData{T, Symbol} where T})(Any, Any, Any, Any, Symbol, T<:Union{DistributedFactorGraphs.ConvolutionObject, DistributedFactorGraphs.FunctorInferenceType}) where {T<:Union{DistributedFactorGraphs.ConvolutionObject, DistributedFactorGraphs.FunctorInferenceType}} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/entities/DFGFactor.jl:58 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraphTypes.jl:313.
WARNING: Method definition (::Type{DistributedFactorGraphs.GenericFunctionNodeData{T, Symbol} where T})(Any, Any, Any, Any, Symbol, T<:Union{DistributedFactorGraphs.ConvolutionObject, DistributedFactorGraphs.FunctorInferenceType}, String) where {T<:Union{DistributedFactorGraphs.ConvolutionObject, DistributedFactorGraphs.FunctorInferenceType}} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/entities/DFGFactor.jl:58 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraphTypes.jl:313.
WARNING: Method definition (::Type{DistributedFactorGraphs.GenericFunctionNodeData{T, Symbol} where T})(Any, Any, Any, Any, Symbol, T<:Union{DistributedFactorGraphs.ConvolutionObject, DistributedFactorGraphs.FunctorInferenceType}, String, Array{Int64, 1}) where {T<:Union{DistributedFactorGraphs.ConvolutionObject, DistributedFactorGraphs.FunctorInferenceType}} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/entities/DFGFactor.jl:58 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraphTypes.jl:313.
WARNING: Method definition (::Type{DistributedFactorGraphs.GenericFunctionNodeData{T, #s54} where #s54<:AbstractString where T})(Any, Any, Any, Any, S<:AbstractString, T<:DistributedFactorGraphs.PackedInferenceType) where {T<:DistributedFactorGraphs.PackedInferenceType, S<:AbstractString} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/entities/DFGFactor.jl:56 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraphTypes.jl:317.
WARNING: Method definition (::Type{DistributedFactorGraphs.GenericFunctionNodeData{T, #s54} where #s54<:AbstractString where T})(Any, Any, Any, Any, S<:AbstractString, T<:DistributedFactorGraphs.PackedInferenceType, String) where {T<:DistributedFactorGraphs.PackedInferenceType, S<:AbstractString} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/entities/DFGFactor.jl:56 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraphTypes.jl:317.
WARNING: Method definition (::Type{DistributedFactorGraphs.GenericFunctionNodeData{T, #s54} where #s54<:AbstractString where T})(Any, Any, Any, Any, S<:AbstractString, T<:DistributedFactorGraphs.PackedInferenceType, String, Array{Int64, 1}) where {T<:DistributedFactorGraphs.PackedInferenceType, S<:AbstractString} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/entities/DFGFactor.jl:56 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraphTypes.jl:317.
WARNING: Method definition getData(DistributedFactorGraphs.DFGFactor{T, S} where S where T) in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:90 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:10.
WARNING: Method definition getData(DistributedFactorGraphs.DFGVariable) in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:91 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:11.
WARNING: Method definition #getData(Any, typeof(DistributedFactorGraphs.getData), DistributedFactorGraphs.DFGVariable) in module DistributedFactorGraphs overwritten in module IncrementalInference.
WARNING: Method definition getSofttype(DistributedFactorGraphs.VariableNodeData) in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:173 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:58.
WARNING: Method definition getSofttype(DistributedFactorGraphs.DFGVariable) in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:176 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:61.
WARNING: Method definition #getSofttype(Any, typeof(DistributedFactorGraphs.getSofttype), DistributedFactorGraphs.DFGVariable) in module DistributedFactorGraphs overwritten in module IncrementalInference.
WARNING: Method definition getfnctype(DistributedFactorGraphs.GenericFunctionNodeData{T, S} where S where T) in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:107 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:99.
WARNING: Method definition getfnctype(DistributedFactorGraphs.DFGFactor{T, S} where S where T) in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:115 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:106.
WARNING: Method definition #getfnctype(Any, typeof(DistributedFactorGraphs.getfnctype), DistributedFactorGraphs.DFGFactor{T, S} where S where T) in module DistributedFactorGraphs overwritten in module IncrementalInference.
WARNING: Method definition getfnctype(T<:DistributedFactorGraphs.AbstractDFG, Symbol) where {T<:DistributedFactorGraphs.AbstractDFG} in module DistributedFactorGraphs at /home/ufechner/.julia/packages/DistributedFactorGraphs/Q6p1P/src/Common.jl:119 overwritten in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/FactorGraph01.jl:111.
WARNING: Method definition #getfnctype(Any, typeof(DistributedFactorGraphs.getfnctype), T<:DistributedFactorGraphs.AbstractDFG, Symbol) where {T<:DistributedFactorGraphs.AbstractDFG} in module DistributedFactorGraphs overwritten in module IncrementalInference.
WARNING: Method definition getVariableInferredDim(DistributedFactorGraphs.VariableNodeData) in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/InferDimensionUtils.jl:48 overwritten at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/InferDimensionUtils.jl:66.
WARNING: Method definition getVariableInferredDim(DistributedFactorGraphs.VariableNodeData, Bool) in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/InferDimensionUtils.jl:48 overwritten at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/InferDimensionUtils.jl:66.
┌ Warning: Replacing docs for `IncrementalInference.getVariableInferredDim :: Union{Tuple{VariableNodeData}, Tuple{VariableNodeData,Bool}}` in module `IncrementalInference`
└ @ Base.Docs docs/Docs.jl:223
WARNING: Method definition lsf(IncrementalInference.FactorGraph) in module IncrementalInference at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/Deprecated.jl:237 overwritten at /home/ufechner/.julia/packages/IncrementalInference/SYZv5/src/Deprecated.jl:290.
[... cut ...]
 [15] run_repl(::REPL.AbstractREPL, ::Any) at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.1/REPL/src/REPL.jl:192
 [16] (::getfield(Base, Symbol("##734#736")){Bool,Bool,Bool,Bool})(::Module) at ./client.jl:362
 [17] #invokelatest#1 at ./essentials.jl:742 [inlined]
 [18] invokelatest at ./essentials.jl:741 [inlined]
 [19] run_main_repl(::Bool, ::Bool, ::Bool, ::Bool, ::Bool) at ./client.jl:346
 [20] exec_options(::Base.JLOptions) at ./client.jl:284
 [21] _start() at ./client.jl:436

```

dehann, 2019-07-21:
hi @ufechner7 , oh thanks for reporting!  

> ERROR: LoadError: InitError: UndefVarError: libwand not defined
   Stacktrace:
   [1] magickgenesis at /home/ufechner/.julia/packages/ImageMagick/vMfoS/src/libmagickwand.jl10

I haven't seen this before and will look into it.   Two things jump out at me: ImageMagick may have some recent changes that we need to look at -- the libwand issue is probably something upstream .

> WARNING: Method definition ...

Also DistributedFactorGraphs (DFG) is major new piece of code we are introducing further upstream (around IncrementalInference.jl, IIF).  This code should bring several significant improvements to the code quality and user experience, but we are working out a few teething problems at the moment.  The plan is to get to IIF v0.8.0 as fast as possible, with possibly fragile behavior in v0.7.x, which is where we are working furiously at the moment.  We have been working on the new code for many months and started the transition about a month ago.  We hope to be finished soon, if you able to hold out for a week or two more?

I hope the new code experience will be a good deal better as we get it done, apologies if the transition is a little bumpy.

IIF v0.6.x should still be stable and was purposefully tagged just before we started the migration to DFG.

dehann, 2020-03-25:
this should be fixed with DFG v0.7, IIF v0.10, RoME v0.6, Caesar v0.5.0.  BTW v0.5.1 will be landing soon with many fixes relating internal consolidation following the general migration on DistributedFactorGraphs.jl

closed 2020-03-25 by dehann