JuliaRobotics/Caesar.jl #382: Odometry disconnect at the end of long chains

mc2922, 2019-08-22:
Odometry chains break towards the end of a longer chain? 

190 poses:
[ex2-1.pdf](https://github.com/JuliaRobotics/Caesar.jl/files/3531960/ex2-1.pdf)
[ex2-2.pdf](https://github.com/JuliaRobotics/Caesar.jl/files/3531961/ex2-2.pdf)

200 poses:
[ex1-1.pdf](https://github.com/JuliaRobotics/Caesar.jl/files/3531955/ex1-1.pdf)
[ex1-2.pdf](https://github.com/JuliaRobotics/Caesar.jl/files/3531956/ex1-2.pdf)

dehann, 2019-08-25:
Depiction of the Bayes tree that associated with this issue.  The short branch has a prior, but that prior information is not properly distributed down the long sibling branch.

[image]

dehann, 2019-08-25:
another anecdotal piece of information is that the problem also occurred on a single long branch.

dehann, 2019-09-02:
From the photo above, the triangle is x169, square is x170.  When looking at the clique low down on right hand branch (x169), the following subgraph is obtained.  Note priors represent the downward belief messages during the final stages of the complete solve.  This is where the disconnect in odo seems to be occurring:

[image]

dehann, 2019-09-02:
All these loose fragments should also be computed in parallel btw.

dehann, 2019-09-03:
Ha, a little more progress.  By looking closer I found that odometry factor x168--x169 is working the wrong way round for some reason,
[image]

 and then found this.  The order of fields x168 and x169 is important here, **but clearly wrong**:
[image]

@GearsAD , do you have any ideas here -- think we spoke a little while back on this but I cannot recall what we decided to do about `fncargvID`?

Values in factor seem fine:
[image]

Oh, and the factor graph here `sfg31bd` stands for sub factor graph inside cliq 31 before the down solve is performed.
[image]

dehann, 2019-09-03:
this does not seem to be present in the original factor graph:
[image]

dehann, 2019-09-03:
right, so part of the problem was that downward results were not being transferred from cliq sub graphs back to main dfg -- this was an oversight in the new CSM code.  However, the issue is not yet fully resolved since downward messages from root into the target clique might have inconsistencies, and I'm looking specifically at that now.

dehann, 2019-10-10:
performance should be better with II v0.7.7 or greater.  Keep open for rigorous testing

dehann, 2020-01-01:
need to retest -- currently on IIF v0.8.3 with many improvements since this issue was logged.

dehann, 2020-02-06:
Will re-evaluate after JuliaRobotics/IncrementalInference.jl#579 is fixed