JuliaRobotics/Caesar.jl #395: [ROS, Decision] How to provide non-Gaussian with vanilla ROS msgs?

dehann, 2019-10-11:
ROS SLAM does not really cater for non-Gaussian.  So what is the best way to be vanilla, but strawberry and blueberry too?

Also related, 
- #227 
- Return types: JuliaRobotics/DistributedFactorGraphs.jl#134
- PPE: JuliaRobotics/DistributedFactorGraphs.jl#149

dehann, 2019-10-11:
https://github.com/jdlangs/RobotOS.jl/blob/master/test/services.jl

dehann, 2019-10-11:
https://github.com/JuliaRobotics/Caesar.jl/blob/36ed37febe916410a293d143c26c65fd1b869e99/examples/ros/ServiceExample.jl#L19

pvazteixeira, 2019-10-12:
See also:

- [geometry_msgs](http://wiki.ros.org/geometry_msgs?distro=melodic)
- [sensor_msgs](http://wiki.ros.org/sensor_msgs?distro=melodic)

dehann, 2019-10-15:
Should we develop our own messages and then also populate the existing vanilla interface with printouts saying that those are the reduced Gaussian-only "wrapper" versions.  That will allow people to try the new message types after trying out the conventional Gaussian only types.  There might be cause to support only new non-Gaussian msg types, but lets start from the familiar stuff and work our way up.  Same amount of development on the Julia side to support ROS msgs anyhow.