JuliaRobotics/Caesar.jl #474: tag v0.4.3

dehann, 2020-01-01:
@JuliaRegistrator register branch=release/v0.4

JuliaRegistrator, 2020-01-01:
Registration pull request created: [JuliaRegistries/General/7414](https://github.com/JuliaRegistries/General/pull/7414)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.3 -m "<description of version>" 25ee8d3d2fa17993e29bd06e725564338e8a9bd6
git push origin v0.4.3
```

dehann, 2020-01-01:
@JuliaRegistrator register branch=release/v0.4

JuliaRegistrator, 2020-01-01:
Registration pull request updated: [JuliaRegistries/General/7414](https://github.com/JuliaRegistries/General/pull/7414)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.3 -m "<description of version>" 1d085e3d654f2a42fd3a554cd25f174b347c76f6
git push origin v0.4.3
```

closed 2020-01-01 by dehann