JuliaRobotics/Caesar.jl #597: tag v0.6.0

dehann, 2020-08-20:
@JuliaRegistrator register branch=release/v0.6

JuliaRegistrator, 2020-08-20:
Registration pull request created: [JuliaRegistries/General/19819](https://github.com/JuliaRegistries/General/pull/19819)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.6.0 -m "<description of version>" 98d12f65eab77b6a1222d8d41d06186ab3ddce6b
git push origin v0.6.0
```

dehann, 2020-08-20:
@JuliaRegistrator register branch=release/v0.6

JuliaRegistrator, 2020-08-20:
Registration pull request updated: [JuliaRegistries/General/19819](https://github.com/JuliaRegistries/General/pull/19819)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.6.0 -m "<description of version>" e5be255a915cac555266d6409266f2b2aee3efd1
git push origin v0.6.0
```

closed 2020-08-20 by dehann