JuliaRobotics/Caesar.jl #721: "UndefVarError: emptyBayesTree not defined" when using manageSolveTree!

fabid, 2021-07-04:
I have been following the documentation, as well as the example regarding RobotOS integration ( https://github.com/JuliaRobotics/Caesar.jl/blob/7b946d951c26faf75945ed4928c05fbf5ce9a479/examples/wheeled/racecar/ros/CarFeedCommonSetup.jl ), and stumbled on this error when attempting a managetSolveTree!

To reproduce
```
using Distributed
using RobotOS, Caesar
Distributed.@everywhere using Caesar
robotslam = SLAMWrapperLocal()
ST = manageSolveTree!(robotslam.dfg, robotslam.solveSettings, dbg=false)
```

I couldn't find in the JuliaRobotics codebase where the emptyBayesTree is defined. Is the example deprecated? Am I missing something?

PS: thank you for this amazing project

dehann, 2021-07-08:
Good day @fabid , ah sorry for the slow reply.  I missed this earlier!

The missing function I think changed to just `BayesTree()` to construct a new empty one.  The excessive name `emptyBayesTree` was deprecated and looks like it slipped through.  

I also could not find the function when searching...  Let dig a little deeper.  Yes the example is a little old, but should not be too much to get it going again.  The idea is that manageSolveTree! should let solves run asynchronously while more nodes are added to the factor graph.  New solves will trigger when enough poses / variables were added (some parameter tweaking there).  Lets say every 10 poses a new solve is triggered.  So after 0 to 10 the first solve starts and the graph can keep growing.  manageSolveTree! will either block the main thread at pose 20 if the solve is still busy, or continue on if the previous solve has completed.  Idea is there will always be up to 20 poses then which have not yet been solved before the process is blocked.  This is very useful when processing ROS bags.  The signalling uses Channels if I remember correctly.

Let me see if I can figure out why this error and will post back.

fabid, 2021-07-08:
Thanks for the feedback! I will see what I can do using `BayesTree()` instead.

In our usecase, we have multiple robots (less than 10) doing robot to landmark and robot to robot relative pose measurementsat low frequency (once a minute for example) that we want to integrate in the factor graph. I would like to have a ros node doing regular solves while subscribing to the robots' odometries and publishing at high (similar to odometry) frequency the localization of all the robots (using factor graph solution + odometry update). I first tried to use the async option for the solve, but ran into issues with pub/sub freeze... And it seemed that `manageSolveTree!` might be a solution, but am happy with any solution enabling this pub/sub without blocking. 
I am still farily new to Julia and the interaction of async with RobotOS is not clear to me yet.