JuliaRobotics/Caesar.jl #863: Upgrade to Manifolds.jl v0.8

Affie, 2022-06-07:


Affie, 2022-06-16:
I’m waiting for fixes on manifolds.jl, should be soon. Then i can run test again just to confirm, but everything should work.

Affie, 2022-06-16:
Don’t know about Caesar yet.

dehann, 2022-06-16:
Okay, i did see Mateusz' PR on Thursday comment for all expmaps, sounds like that's where it's at.  Hopefully Manifolds tags that patch fix soon.  Thanks.

dehann, 2022-06-28:
I have this as the conclusion now that Manifolds.jl v0.8.11 has landed:
- https://github.com/JuliaManifolds/Manopt.jl/discussions/90#discussioncomment-3029601

Affie, 2022-06-28:
> This isn't really an epic, I'm going to move to project board (hope that is okay?):

Yes, it's okay, just needed it tracked somehow, but I do think it falls well in the epic category, under the "necessary maintenance" initiative.

Affie, 2022-06-29:
Caesar and RoMEPlotting seem to work without any changes for Manifolds.jl v0.8.
Should we:
- [ ] do a patch release on them, keeping old compat and just adding new
- [ ] do a minor release, keeping old compat and just adding new
- [ ] do a minor release, remove previous compat

dehann, 2022-06-30:
I'd say just do a 1. a patch release, and it's okay to pull min compat bound for MJL up to v0.8.11

Affie, 2022-07-01:
Everything passes tests, with just Caesar.jl left to tag. Let's try Caesar master out for a bit before tagging to make sure it's not missing something since I didn't test anything myself.

closed 2022-07-01 by Affie