JuliaRobotics/Caesar.jl #889: tag v0.13.3

dehann, 2022-07-25:
@JuliaRegistrator register branch=release/v0.13

JuliaRegistrator, 2022-07-25:
Registration pull request created: [JuliaRegistries/General/64957](https://github.com/JuliaRegistries/General/pull/64957)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.13.3 -m "<description of version>" 53d843a4347bb7d9eca3fd0f0d22356879192f97
git push origin v0.13.3
```

dehann, 2022-07-25:
@JuliaRegistrator register branch=release/v0.13

JuliaRegistrator, 2022-07-25:
Registration pull request updated: [JuliaRegistries/General/64957](https://github.com/JuliaRegistries/General/pull/64957)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.13.3 -m "<description of version>" de917865d1e169624d9b0398ea317cc96791e387
git push origin v0.13.3
```

closed 2022-07-25 by dehann