JuliaRobotics/Caesar.jl #910: update ROS1 docs on bag reader and writer not needing roscore, but live traffic does

dehann, 2022-09-18:


dehann, 2022-09-19:
xref:
- https://github.com/NavAbility/BashingMapper/pull/26

dehann, 2023-03-27:
HI @Affie , could you link perhaps the Python script you've been using please?  The one that can read bag files but does not need the rest of ROS...

Affie, 2023-03-27:
```julia
using PyCall
py"""
# needs rosbags installed in python `pip install rosbags`
from rosbags.rosbag1 import Reader
from rosbags.serde import deserialize_ros1, deserialize_cdr, ros1_to_cdr
"""
#setup the rosbag reader 
reader = py"filename.bag')"
# open rosbag
reader.open()
# list of topics
topics = reader.topics
# reading like this
for (connection, timestamp, rawdata) in reader.messages()
    msg = py"deserialize_ros1($(Vector{UInt8}(rawdata)),$(connection[3]))"
    @show msg
    #do something with msg
end
```